MicrosoftDocs/windows-driver-docs #1868: Attestation Signing DOES Work For Server -- Always Has, Still Does

PeterGV, 2019-11-01:
In the NOTE that appears in the colored box at the top of the page, the forth bullet point is bolded, and is reads:

**An attestation signed driver will only work for Windows 10. It will not work for other versions of Windows, such as Windows 8.1, Windows 7, or any Windows Server versions.**

This is helpful, except that the part that says "or any Windows Server versions" is blatantly incorrect.

I've corrected this on other doc pages in the past: Attestation Signing does indeed work on Windows Server.

Please can we route-out all vestiges of this misinformation?  Pretty please?  It's problematic to point an OEM or IHV at this page (which does a pretty nice job describing Attestation Signing) when you have to tell them "but ignore the part about Windows Server that's in BOLD at the top of the page.  That decision was reversed at the last minute and that policy never really did take effect.  Yes.  Just trust me."

Peter

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: ed287852-04b7-1569-fd51-2640cfa24f57
* Version Independent ID: a8ce732f-5b41-f1ce-e37d-a7c2794546a4
* Content: [Attestation signing a kernel driver for public release - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/attestation-signing-a-kernel-driver-for-public-release)
* Content Source: [windows-driver-docs-pr/dashboard/attestation-signing-a-kernel-driver-for-public-release.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/dashboard/attestation-signing-a-kernel-driver-for-public-release.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @EliotSeattle
* Microsoft Alias: **eliotgra**

EliotSeattle, 2019-11-04:
Hi @PeterGV 
Thanks for catching this. My understanding is that attestation signing will not work on Windows Server 2016, so I have added that phrase to make it clear.
Eliot

closed 2019-11-04 by EliotSeattle

PeterGV, 2019-11-04:
Good fix, Eliot.  Absolutely correct and better than what I had proposed.  Many thanks, as always.