MicrosoftDocs/windows-driver-docs #1876: Code 55 and 56 and possibly more codes are missing from the list.

perhof, 2019-11-04:
What does code 55 and code 56 mean?
They have been reported in later versions of Windows 10.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 6b4cb992-5a3e-6964-4828-ea8e9c782fe4
* Version Independent ID: 802a2794-c7c8-d96f-e276-ece58f8521e7
* Content: [Device Manager Error Messages - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/device-manager-error-messages#feedback)
* Content Source: [windows-driver-docs-pr/install/device-manager-error-messages.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/install/device-manager-error-messages.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

RAJU2529, 2019-11-06:
I knew about error code 55 and 56.

perhof, 2020-02-20:
@Raju2529 Thanks for sharing

tedhudek, 2020-03-04:
Thanks for the report @perhof.  Docs are now up for 55 and 56 (and a special bonus, 57!)

https://docs.microsoft.com/en-us/windows-hardware/drivers/install/cm-prob-console-locked

closed 2020-03-04 by tedhudek