MicrosoftDocs/windows-driver-docs #1877: "Concepts for all driver developers link" is broken

nikolodus, 2019-11-05:
The link for this page in the navigation bar on the left points to this page: https://docs.microsoft.com/en-us/
Please resolve this issue.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: db4c3281-0648-3a6f-ca3b-479031099b1e
* Version Independent ID: b10a50cf-766d-2c14-16bc-bce71c5e1ef5
* Content: [Driver stacks - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/gettingstarted/driver-stacks#feedback)
* Content Source: [windows-driver-docs-pr/gettingstarted/driver-stacks.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/gettingstarted/driver-stacks.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @EliotSeattle
* Microsoft Alias: **eliotgra**

nikolodus, 2019-11-05:
Looking at the source, it seems to me that the link should point to this file/page instead: https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/gettingstarted/concepts-and-knowledge-for-all-driver-developers.md

EliotSeattle, 2019-11-05:
Thanks, @nikolodus 
I've made the fix and pushed it to the master branch. It should be live shortly.
Appreciate the help.
Eliot

closed 2019-11-05 by EliotSeattle