MicrosoftDocs/windows-driver-docs #1878: Private virtual address space for ear user mode process

AboliKhare, 2019-11-05:
I would like to get clarification regarding how private virtual address space of one process being isolated from another process for a 64 bit windows uses whole 8TB for one process as mentioned in diagram and such 8 TBs are used for other user space processes but as a whole we have only alloted 8TB for user space?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: ebe87d57-198f-66f4-69ff-6fdbc15eb214
* Version Independent ID: 880a9327-3f4f-83eb-c0f7-030588344330
* Content: [Virtual address spaces - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/gettingstarted/virtual-address-spaces#feedback)
* Content Source: [windows-driver-docs-pr/gettingstarted/virtual-address-spaces.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/gettingstarted/virtual-address-spaces.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @EliotSeattle
* Microsoft Alias: **eliotgra**

tedhudek, 2020-07-17:
Apologies for the delay. Typically a user-mode process would use far less than its 8 TB VA allocation.  The VA space is unique per-process, with the system managing the mappings into physical address space to avoid conflicts. If this is still an issue and this doesn't address your question, please re-open with more detail

closed 2020-07-17 by tedhudek

AboliKhare, 2020-07-17:
@tedhudek thanks for the answer.