MicrosoftDocs/windows-driver-docs #1880: Discuss characteristics of the 3 kernel debug methods

rtischer8277, 2019-11-06:
Reading this page leaves me no better off regarding what to do in order to solve my software problem. What does the kernel code look like on the screen for each method? Is source shown for any of them? I would save a lot of code experiment and learning time if there was a general description of each's characteristics as compared to the other.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 97eed641-8cd3-4359-aee7-2f14acd28b24
* Version Independent ID: c0c6d81e-4487-cd62-9dcd-9ce3b14f98e0
* Content: [Setting Up Local Kernel Debugging of a Single Computer Manually - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/setting-up-local-kernel-debugging-of-a-single-computer-manually)
* Content Source: [windows-driver-docs-pr/debugger/setting-up-local-kernel-debugging-of-a-single-computer-manually.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/setting-up-local-kernel-debugging-of-a-single-computer-manually.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

RAJU2529, 2019-11-11:
What software problems you have experienced. this article about debug the Kernel device driver

rtischer8277, 2019-11-11:
Several times now my project's dev progress has been halted by not being able to debug into underlying Microsoft `socket` and `bind` functions. My first encounter was [how to determine socket interface type](https://stackoverflow.com/questions/57732163/how-to-determine-socket-interface-type) "Since I couldn't debug bind()". My second encounter was [Binding an ipv6 slaac address to windows IP Stack error 10049](https://stackoverflow.com/questions/58663136/binding-an-ipv6-slaac-address-to-windows-ip-stack-error-10049). And I am on my 3rd encounter now which is [Unable to GetUnicastIpAddressEntry after CreateUnicastIpAddressEntry](https://stackoverflow.com/questions/58702391/unable-to-getunicastipaddressentry-after-createunicastipaddressentry) which involves low-level windows socket API calls.

In each case, if I had the ability to debug farther down into the calls, I would have been able to solve the problem and move on. Asking StackOverflow for answers to these low-level call issues is futile, which is why I was exploring kernel-level debuggers. The doc page I was responding to merely described how to use 3 debuggers in kernel mode. It offered zero information on whether they had the potential to let me debug my problem, like whether debugging would show me the source code or not.

Especially now with the transition from ipv4 to ipv6, knowing what is going on inside `socket()`, `bind()` and `listen()` __impl_*** functions is essential.

RAJU2529, 2019-11-11:
@rtischer8277. I thought of you have  normal software program problems. but your issues to  related to developing drivers, So it's better option post a query in visual studio community or forums. In that forum, many experts are there, they will try to solve all your doubts.
I hope you can understand

rtischer8277, 2019-11-11:
I am not developing drivers, but if you have a link to that forum, I'll take a look. Maybe you are saying the `socket()`, `bind()` and `listen()` are referred to as drivers. Is that true? That would explain a lot.

DOMARS, 2019-11-11:
@rtischer8277 - I would post on this forum:
•	[Windows Hardware WDK and Driver Development](https://social.msdn.microsoft.com/Forums/windowsdesktop/en-US/home?forum=wdk)

RAJU2529, 2019-11-11:
@rtischer8277. sorry I don't have those knowledge.

rtischer8277, 2019-11-12:
Thanks all for the guidance. I found this page which describes the debuggers in more detail. If after studying these and I still need advice, I'll re-post on the WDK forum: [Debugging Tools for Windows (WinDbg, KD, CDB, NTSD)](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/)

DOMARS, 2019-11-13:
@rtischer8277 - Good luck with your networking debugging work and thanks for the interest in the Windows Debugger docs.

closed 2019-11-13 by DOMARS