MicrosoftDocs/windows-driver-docs #1881: Windows 10 Drivers

ayousifboe, 2019-11-06:
I am trying to configure drivers to work without Test mode on Windows 10 for a tool I'm developing. Most documentations seems to suggest an EV cert is needed but this seems more unnecessary for what I'm trying to do. I'm not planning to distribute the drivers on Windows update. I was hoping to digitally sign the drivers myself but that seems insufficient. I want to be able to test things outside of test mode. What is the simplest solution to this problem? Thank you.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 1eb19d33-8f4c-f379-b645-938a746c5615
* Version Independent ID: 8efb9fb4-1e9a-1a8a-994b-450495449e15
* Content: [Driver Signing Policy - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/kernel-mode-code-signing-policy--windows-vista-and-later-#feedback)
* Content Source: [windows-driver-docs-pr/install/kernel-mode-code-signing-policy--windows-vista-and-later-.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/install/kernel-mode-code-signing-policy--windows-vista-and-later-.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2019-11-06:
Thanks @ayousifboe for the feedback.  So you have a driver package that you do not want to test sign and you do not want to submit to the dashboard but you want to be able to distribute internally?

ayousifboe, 2019-11-06:
Yes that is correct.

tedhudek, 2019-11-06:
Ok, let me investigate a bit internally and I'll circle back.

tedhudek, 2019-11-07:
Hey Aziz, I'm still working on integrating this into the docs, but try this (assuming it's a kernel mode driver):

1. [Register for the hw program](https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/register-for-the-hardware-program).
2. [Sign appropriate agreements](https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/hardware-dashboard-faq).
3. Upload codesign certificates to Hardware Dev Center.
2.	Sign drivers locally, using non-EV codesign cert. 
3.	Package drivers in a CAB, sign CAB using above codesign cert.
4.	Submit CAB to HDC for signing.
5.	HDC returns submitted drivers now signed by MSFT.
6.	Distribute driver internally, no test mode required.

ayousifboe, 2019-11-07:
Thank you this is helpful. Quick questions- So when you mean non-EV code sign cert, can these be certs I generate myself with openssl, for example? And, for signing appropriate agreements, what if I am not an authority to sign contracts on behalf of my organization?

tedhudek, 2019-11-08:
EV, and non-EV codesigning certificates can be purchased from any of the usual PKI vendors.  Regarding signing contracts, I'm not sure.

ayousifboe, 2019-11-08:
So just to confirm, code singing certs have to be purchased? They cannot be created locally on my own?

rodwiddowson, 2019-11-08:
I assume that your have tried putting the public part of your key signing cert (or a if you really like the pain, something in the PKICS chain) into the keystore on your deployment machines?  I cannot remember which store it goes into but some experimentation in a more would help,

This has worked (for a boot start FSD driver) for me in recent memory, but I’ve not tested of late.  

Also note that things which are needed in preboot (like windbg DLLS) need extra special signing and not even an EV cert will help.

Did you also ask in NTDEV?

Or have i got the wrong end of the stick? In general I eschew everything to do withX509 and in particular driver signing...

closed 2020-03-09 by tedhudek