MicrosoftDocs/windows-driver-docs #1883: Bad Link

KappaRan, 2019-11-07:
https://go.microsoft.com/fwlink/p/?linkid=62651

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 9d5f648a-f925-7bca-7bad-b97fded54baf
* Version Independent ID: e6a51140-c16d-8096-7888-2536b23a3ebc
* Content: [Installing Device Metadata Packages from WMIS - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/installing-device-metadata-packages-from-wmis)
* Content Source: [windows-driver-docs-pr/install/installing-device-metadata-packages-from-wmis.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/install/installing-device-metadata-packages-from-wmis.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

RAJU2529, 2019-11-11:
@wdkbot. today I will replace the exact link afternoon, by creating new PR

RAJU2529, 2019-11-11:
@wdkbot. Just now I created PR #1897

tedhudek, 2019-11-11:
Fixed in #1897.  Thanks for the report @KappaRan

closed 2019-11-11 by tedhudek