MicrosoftDocs/windows-driver-docs #1887: Missing descriptions for arg1 = 0x1240 and 0x1241

paullev, 2019-11-07:
I'm seeing these bugcheck sub-types and would like to know what they mean. 

Regards,
Paul

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: de955911-dfbf-33b3-778a-767a5a578a0d
* Version Independent ID: 23fb5cbd-5f7d-f7ca-1622-5885020966c2
* Content: [Bug Check 0x1A MEMORY_MANAGEMENT - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-0x1a--memory-management#feedback)
* Content Source: [windows-driver-docs-pr/debugger/bug-check-0x1a--memory-management.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/bug-check-0x1a--memory-management.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

paullev, 2019-11-07:
MEMORY_MANAGEMENT (1a)
    # Any other values for parameter 1 must be individually examined.
Arguments:
Arg1: 0000000000001240, The subtype of the bugcheck.
Arg2: ffff898816026ad0
Arg3: fffffee482baefa0
Arg4: 0000200000000080

paullev, 2019-11-07:
MEMORY_MANAGEMENT (1a)
    # Any other values for parameter 1 must be individually examined.
Arguments:
Arg1: 0000000000001241, The subtype of the bugcheck.
Arg2: ffffbc02a9d63900
Arg3: fffffef28382c7d8
Arg4: 0000200000000080

DOMARS, 2020-02-04:
@paullev - I worked with the dev team to add descriptions for 1240 and 1241, so closing this issue. Thanks for your interest in improving the Windows Driver docs.

closed 2020-02-04 by DOMARS