MicrosoftDocs/windows-driver-docs #1888: Redundancies of Tool Location

justinmcbride, 2019-11-08:
The following sentence is repeated twice on this page:

`You can also manually run the Stampinf tool, which is located in the bin subdirectory of the WDK.`. Just FYI.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 91ae9573-bdc0-52c5-ebca-490b8ff599f8
* Version Independent ID: 36db20b1-e742-95b8-9147-163fbc0ba0e0
* Content: [Using INX Files to Create INF Files - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/wdf/using-inx-files-to-create-inf-files#feedback)
* Content Source: [windows-driver-docs-pr/wdf/using-inx-files-to-create-inf-files.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/wdf/using-inx-files-to-create-inf-files.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

RAJU2529, 2019-11-11:
@wdkbot. I created new PR #1896

tedhudek, 2019-11-11:
Fixed by #1896.  Thanks for the report.

closed 2019-11-11 by tedhudek