MicrosoftDocs/windows-driver-docs #1889: New bugcheck 239

LuigiBruno, 2019-11-08:
I'm troubleshooting a CRITICAL_PROCESS_DIED bugcheck (no dump file available), only a Kernel-Power event (ID 41) on a Windows Server machine: the detailed information for the event report a "BugcheckCode 239" event data.
Is that a new bugcheck not documented yet?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: ddc632b2-f4e1-b784-dd1e-878feac077f9
* Version Independent ID: 7a6866f4-3333-f2d1-86d7-dbb3d00925e0
* Content: [Bug Check Code Reference - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-code-reference2#feedback)
* Content Source: [windows-driver-docs-pr/debugger/bug-check-code-reference2.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/bug-check-code-reference2.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2019-11-08:
@LuigiBruno - Can you provide the first part of the !analyze output that shows the four parameters? Also can you verify the hex value of this stop code? Also we should check if this is VMWare related.

LuigiBruno, 2019-11-08:
> @LuigiBruno - Can you provide the first part of the !analyze output that shows the four parameters? Also can you verify the hex value of this stop code? Also we should check if this is VMWare related.

Hi @DOMARS.
There was no dump file, therefore I couldn't use WinDbg.
Here's the information from the "System" log in the Windows Event Viewer

BugcheckCode 239
BugcheckParameter1 0xffff930e74673340
BugcheckParameter2 0x0
BugcheckParameter3 0x0
BugcheckParameter4 0x0

RAJU2529, 2019-11-09:
@LuigiBruno. Enable crash control using registry and windows settings. For more informations  visit below sites
**https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/enabling-a-kernel-mode-dump-file**

**https://docs.microsoft.com/en-us/windows/client-management/generate-kernel-or-complete-crash-dump**

**https://www.gfi.com/support/products/How-to-collect-user-mode-crash-dumps-with-Windows-Error-Reporting**

**https://carrona.org/bsodreg.html**

RAJU2529, 2019-11-09:
@wdkbot. Sir, can I add some reference links to this article by creating new PR?

LuigiBruno, 2019-11-15:
@DOMARS and @RAJU2529, it should be 0x000000EF (239 in decimal), definitively a CRITICAL_PROCESS_DIED bug check.

DOMARS, 2020-02-25:
@LuigiBruno - We have documented the CRITICAL_PROCESS_DIED bug check here.

https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-0xef--critical-process-died

@RAJU2529 - We have links to these two topics, which should be sufficient pointers for a dev to find what they need to fix the issues in their code.

- Crash dump analysis using the Windows debuggers (WinDbg)

- Analyzing a Kernel-Mode Dump File with WinDbg

For general discussion of stop codes, the OSR or WDK forums are the best place for that:

[Windows Hardware WDK and Driver Development](https://social.msdn.microsoft.com/Forums/windowsdesktop/en-US/home?forum=wdk)

closed 2020-02-25 by DOMARS