MicrosoftDocs/windows-driver-docs #1890: Description and parameters

LuigiBruno, 2019-11-08:
What does this bug check indicate? Which could be the values for its parameters?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 104442f4-9a8a-ee2b-a035-5723ffdae0ef
* Version Independent ID: 8dcd0a0e-feba-4403-9308-277c45a71fd7
* Content: [Bug Check 0x8 IRQL_NOT_DISPATCH_LEVEL - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-0x8--irql-not-dispatch-level#feedback)
* Content Source: [windows-driver-docs-pr/debugger/bug-check-0x8--irql-not-dispatch-level.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/bug-check-0x8--irql-not-dispatch-level.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2020-02-27:
@LuigiBruno  - Thanks for the input, that we should add additional information on Bug Check 0x8. 

We use customer input like this as well as data on how often the bug checks occur, to determine how we allocate our limited documentation team resources.

I'm not able to provide any additional information on this stop code.

This stop code doesn't currently occur on a frequent basis, but we will continue to monitor it and the other stop codes. If it becomes more active in the future, we can consider further work.

Thanks for your interest in improving the stop code documentation.

closed 2020-02-27 by DOMARS

LuigiBruno, 2020-03-20:
> @LuigiBruno - Thanks for the input, that we should add additional information on Bug Check 0x8.
>
> We use customer input like this as well as data on how often the bug checks occur, to determine how [...]

OK @DOMARS. Thank you.