MicrosoftDocs/windows-driver-docs #1894: After boot into the system, the use of touch pen point does not disappear;

VensonHe, 2019-11-11:
[Enter feedback here]
 Condition:
1,Platform: CometLake -U, 
2,Windows: x64-1903
3,TP IC: Goodix7380
Issue:
Boot into system, Use the touch pen,but the pen always on the screen
We have try something test:
1, Closed the faststartup function, issue will gone 
2, Reboot and test, issue will gone
3, Reduce the system version to RS5, issue will gone.

so, Can you give me some idea to fix it on 19H1?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 10167325-2086-129e-2936-4fd35bd86d5b
* Version Independent ID: 8785a8a9-16d6-7dfb-83c6-13a2d7f2d9c7
* Content: [Distinguishing Fast Startup from Wake-from-Hibernation - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/kernel/distinguishing-fast-startup-from-wake-from-hibernation)
* Content Source: [windows-driver-docs-pr/kernel/distinguishing-fast-startup-from-wake-from-hibernation.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/kernel/distinguishing-fast-startup-from-wake-from-hibernation.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

RAJU2529, 2019-11-11:
@VensonHe. Sir it's better to post your query or problems in https://answers.microsoft.com

VensonHe, 2019-11-11:
Dear sir
Thanks for your quick response.
OK, I will ask 'him' for help.

closed 2019-11-11 by VensonHe