MicrosoftDocs/windows-driver-docs #1900: Question about MBIM structure

CeciAc, 2019-11-14:
Hi team, 
I moved this from here: https://github.com/MicrosoftDocs/windows-driver-docs.zh-cn/issues/7

@XXXXzzzz000 commented 2 hours ago •

where can i got the structure define of MBIM_MS_SET_UICC_TERMINAL_CAPABILITY?
I only can understand : this payload is a struct array, but i can't get any information of this struct array
Thanks.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: c17a35ca-60a6-fd9f-e171-8413f8ea4604
* Version Independent ID: 7516b9db-e539-19d7-9be6-1140f6141404
* Content: [MB low level UICC access - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/network/mb-low-level-uicc-access)
* Content Source: [windows-driver-docs-pr/network/mb-low-level-uicc-access.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/network/mb-low-level-uicc-access.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @duncanmacmichael
* Microsoft Alias: **dumacmic**

RAJU2529, 2019-11-16:
@CeciAc. I think you can get here
https://docs.microsoft.com/en-us/windows-hardware/drivers/network/mb-interface-model

XXXXzzzz000, 2019-11-18:
such as this page https://docs.microsoft.com/en-us/windows-hardware/drivers/network/mb-low-level-uicc-access
in `MBIM_MS_ATR_INFO`
i can't get any info about AtrData's struct, how can i parse it?

tedhudek, 2021-02-24:
Thanks for the feedback and apologies for the delay.  I see MBIM_MS_ATR_INFO documented on the page in question. If you still have questions about using it, you can try the forums:

* [Windows Hardware WDK and Driver Development](https://social.msdn.microsoft.com/Forums/windowsdesktop/en-US/home?forum=wdk)
* [OSR's ntdev List](http://www.osronline.com/showlists.cfm?list=ntdev)

closed 2021-02-24 by tedhudek