MicrosoftDocs/windows-driver-docs #1902: screen shot of support request doesn't match current website

sarman1998, 2019-11-14:
Either that or I just can't find the correct place.  The text in the Note, also wasn't helpful.
Does "inside the Partner Center" mean in the Hardware program?  I tried clicking on the Home button in the left panel of the web page, but from the home screen of the partner center, the help and support drop down menu doesn't allow you to submit a support request unless you expand one of the programs, like Hardware.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: c95431c9-6c25-4867-ac56-fdfef48c71f3
* Version Independent ID: 266383ed-2aa4-6c53-d8e0-b965aa9684f5
* Content: [Driver flighting - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/driver-flighting#feedback)
* Content Source: [windows-driver-docs-pr/dashboard/driver-flighting.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/dashboard/driver-flighting.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @EliotSeattle
* Microsoft Alias: **eliotgra**

tedhudek, 2020-11-10:
TODO: review screenshot and note in section https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/driver-flighting#signing-up-for-driver-flighting

tedhudek, 2020-11-16:
Hey @sarman1998 thanks for the feedback and apologies it took so long to get back to you.

I was unable to find a contact to help with your question.  Hopefully you got it resolved by now?  If so, were there any gotchas you think we should add to https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/driver-flighting#signing-up-for-driver-flighting?

Closing this out for now but I'll still monitor it.

closed 2020-11-16 by tedhudek