MicrosoftDocs/windows-driver-docs #1903: Link for BCD Boot Options Reference is incorrect

rossmpersonal, 2019-11-15:
Link for BCD Boot Options Reference is incorrect. Please fix this.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 568925fd-87a5-72dc-7ce3-171839aa7342
* Version Independent ID: a44b76f6-a21e-f091-8657-c89c3cdd0ed9
* Content: [Editing Boot Options - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/devtest/editing-boot-options#feedback)
* Content Source: [windows-driver-docs-pr/devtest/editing-boot-options.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/devtest/editing-boot-options.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @EliotSeattle
* Microsoft Alias: **eliotgra**

RAJU2529, 2019-11-16:
@wdkbot.  Just now i created PR  #1904 
but i don't know, why all checks are failed, so, please correct it, if you  have free time.

EliotSeattle, 2020-04-01:
Thanks for pointing this out. The link is correct, but the link text was not. I've made the change and checked it into the live branch. It should be visible very shortly.
Eliot

closed 2020-04-01 by EliotSeattle