MicrosoftDocs/windows-driver-docs #1906: Parameter 1 = 0x1d: what does it mean?

LuigiBruno, 2019-11-19:
I've troubleshooted this bugcheck with the above value for parameter 1 and the below information provided by the !analyze -v command

PROCESS_NAME:  firefox.exe
ERROR_CODE: (NTSTATUS) 0xc0000409 - The system detected an overrun of a stack-based buffer in this application. This overrun could potentially allow a malicious user to gain control of this application.
EXCEPTION_CODE_STR:  c0000409
EXCEPTION_PARAMETER1:  000000000000001d

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: f1073c44-369a-9365-a2a7-ea5b064fb1af
* Version Independent ID: 28187f46-abb9-c7cb-9b22-ddd4a2ec9d37
* Content: [Bug Check 0x139 KERNEL_SECURITY_CHECK_FAILURE - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-0x139--kernel-security-check-failure#feedback)
* Content Source: [windows-driver-docs-pr/debugger/bug-check-0x139--kernel-security-check-failure.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/bug-check-0x139--kernel-security-check-failure.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2020-02-25:
@LuigiBruno - At this point any additional returned codes are for internal use, so we won't be documenting those. I will let you know if anything changes.

closed 2020-02-25 by DOMARS

LuigiBruno, 2020-03-20:
> @LuigiBruno - At this point any additional returned codes are for internal use, so we won't be documenting those. I will let you know if anything changes.

OK @DOMARS. Thank you.