MicrosoftDocs/windows-driver-docs #1908: 0x6001 status parameter meaning

niderost, 2019-11-20:
I'm debugging a blue screen with stop code 0x0000001A, parameter 1 of 0x0000000000006001 and parameter 2 of 0xFFFFFFFFC0000010. From this page I conclude that 0xFFFFFFFFC0000010 is the "status" returned. It would be helpful if this page gave some background as to what the status returned means / what the specific value 0xFFFFFFFFC0000010 indicates.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: de955911-dfbf-33b3-778a-767a5a578a0d
* Version Independent ID: 23fb5cbd-5f7d-f7ca-1622-5885020966c2
* Content: [Bug Check 0x1A MEMORY_MANAGEMENT - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-0x1a--memory-management)
* Content Source: [windows-driver-docs-pr/debugger/bug-check-0x1a--memory-management.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/bug-check-0x1a--memory-management.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

niderost, 2019-11-20:
I found this page with NTSTATUS values:
https://docs.microsoft.com/en-us/openspecs/windows_protocols/ms-erref/596a1078-e883-4972-9bbc-49e60bebca55
Now my hypothesis is that the memory store component does an internal system call (maybe to retrieve a paged out file?) and parameter 2 indicates that that internal system call returned an error (STATUS_INVALID_DEVICE_REQUEST).
It would have been great if this page contained information that I could use to confirm of falsify my hypothesis, i.e. a link to a page describing what the "memory store component" is, a more detailed description of what is meant by the "returned status" (which component is returning this status, the memory component? Or another component that is called by the memory component? If it is the memory component, what other component is it returning this status to?), and a link to a page enumerating the meaning of possibly returned status values (e.g. a link to the NTSTATUS values page if indeed a NTSTATUS value is returned.)

DOMARS, 2020-02-04:
@niderost - I worked with the dev team to add descriptions for 6001, so closing this issue. 
For additional assistance with your troubleshooting, I would recommend this group: 

•	[Windows Hardware WDK and Driver Development](https://social.msdn.microsoft.com/Forums/windowsdesktop/en-US/home?forum=wdk)

Thanks for your interest in improving the Windows Driver docs.

closed 2020-02-04 by DOMARS