MicrosoftDocs/windows-driver-docs #1909: Reference to a non-existing item

ern0, 2019-11-20:
"When you install Visual Studio 2019, select the »Desktop development with C++ workload«." There is no such. The closest one is "Windows Desktop Application".

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 5930505f-6810-b2a2-86b5-3c678aae7f89
* Version Independent ID: 28315cf8-f06f-23ad-2088-9d51d08efe0d
* Content: [Download the Windows Driver Kit (WDK) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/download-the-wdk#feedback)
* Content Source: [windows-driver-docs-pr/download-the-wdk.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/download-the-wdk.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

ern0, 2019-11-20:
Version:
 Microsoft Visual Studio Community 2019
 Version 16.3.9
 VisualStudio.16.Release/16.3.9+29509.3

EliotSeattle, 2019-11-20:
Bill, can you look at this quickly?

billlattimer, 2019-11-20:
@EliotSeattle Reviewing now.

billlattimer, 2020-01-08:
AB#146084

billlattimer, 2020-02-28:
@ern0, the option will appear in the Visual Studio Installer rather than in Visual Studio. You can open the installer directly from your menu, or when you go to "Create a new project" you'll see a prompt for "Install more tools and features" which will open Visual Studio Installer for you. 

[image]

closed 2020-02-28 by billlattimer