MicrosoftDocs/windows-driver-docs #1910: Visual Studio Installer has no such tab

ern0, 2019-11-20:
"In Visual Studio Installer, on the Workloads tab, under Installation Details, expand Universal Windows Platform development. " There are no Workloads tab, only "Installed" and "Available".

Version: Visual Studio Community 16.3.9

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 5930505f-6810-b2a2-86b5-3c678aae7f89
* Version Independent ID: 28315cf8-f06f-23ad-2088-9d51d08efe0d
* Content: [Download the Windows Driver Kit (WDK) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/download-the-wdk#feedback)
* Content Source: [windows-driver-docs-pr/download-the-wdk.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/download-the-wdk.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2019-11-20:
@billlattimer Can you look at this and #1909 please?

billlattimer, 2019-11-20:
@EliotSeattle reviewing both of these now.

naveenimmadisetti, 2019-11-26:
Not able to see the project templates for drivers in VS2019  even though wdk is reinstalled.

billlattimer, 2020-02-28:
@ern0, as with your other issue, the option will appear in the Visual Studio Installer rather than in Visual Studio. You can open the installer directly from your menu, or when you go to "Create a new project" you'll see a prompt for "Install more tools and features" which will open Visual Studio Installer for you. 

[image]

closed 2020-02-28 by billlattimer