MicrosoftDocs/windows-driver-docs #1914: Parameters and their meanings

LuigiBruno, 2019-11-20:
Add the list of the parameters for this bugcheck and the meaning of their values.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 98c4748a-e828-a6ed-94ea-f25e877531a5
* Version Independent ID: 46d7a626-fef1-02e9-9dc7-a47f78ef536b
* Content: [Bug Check 0x3D INTERRUPT_EXCEPTION_NOT_HANDLED - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-0x3d--interrupt-exception-not-handled#feedback)
* Content Source: [windows-driver-docs-pr/debugger/bug-check-0x3d--interrupt-exception-not-handled.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/bug-check-0x3d--interrupt-exception-not-handled.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2020-02-27:
@LuigiBruno  - Thanks for the input, that we should add additional information on Bug Check 0x3D. 

We use customer input like this as well as data on how often the bug checks occur, to determine how we allocate our documentation team resources.

LuigiBruno, 2020-03-04:
> @LuigiBruno - Thanks for the input, that we should add additional information on Bug Check 0x3D.
> 
> We use customer input like this as well as data on how often the bug checks occur, to determine how we allocate our documentation team resources.

Thank you @DOMARS.

DOMARS, 2020-12-22:
@LuigiBruno - I looked at the  INTERRUPT_EXCEPTION_NOT_HANDLED and I would need to work with the dev team to provide more detail on this error. I have created and internal work item to investigate as time allows. The developer may want to look their interrupt related code and the .ecxr (Display Exception Context Record) debugger command may be useful. Thanks for your interest in the debugger stop codes.

closed 2020-12-22 by DOMARS

LuigiBruno, 2021-01-18:
> @LuigiBruno - I looked at the INTERRUPT_EXCEPTION_NOT_HANDLED and I would need to work with the dev team to provide more detail on this error. I have created and internal work item to investigate as [...]

Thanks you @DOMARS.