MicrosoftDocs/windows-driver-docs #1917: BugCheck 0xC0000244 is caused by the CrashOnAuditFail feature

jokezone, 2019-11-21:
BugCheck 0xC0000244 occurred on one of our servers recently, but that code is not listed in this document.

!analyze -v output returns:

Unknown bugcheck code (c0000244)
Unknown bugcheck description
Arguments:
Arg1: ffffffffc0000188
Arg2: 0000000000000000
Arg3: 0000000000000000
Arg4: 0000000000000000

EXCEPTION_CODE: (NTSTATUS) 0xc0000244 - {Audit Failed}  An attempt to generate a security audit failed.

This led me to these articles identifying the root cause as the **CrashOnAuditFail** feature:

https://support.microsoft.com/en-us/help/2346270/you-receive-a-stop-error-0xc0000244-audit-failed-error-message-during

https://docs.microsoft.com/en-us/previous-versions/windows/it-pro/windows-2000-server/cc963220(v=technet.10)

Is it safe to assume BugCheck 0xC0000244 is caused only by the CrashOnAuditFail feature being triggered? If so, should this bugcheck code be added to this document?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: ddc632b2-f4e1-b784-dd1e-878feac077f9
* Version Independent ID: 7a6866f4-3333-f2d1-86d7-dbb3d00925e0
* Content: [Bug Check Code Reference - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-code-reference2)
* Content Source: [windows-driver-docs-pr/debugger/bug-check-code-reference2.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/bug-check-code-reference2.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2020-02-04:
@jokezone - Apologies for the delay here as this is a difficult question. I do not believe this to be a stop code in Windows. It may be a valid error log message and value, but not a stop code as in a kernel bug check that Windows created. If this is still an issue, can you clarify how you ran !analyze. When you invoked it and what specific command you provided? Is this something you can still replicate?

DOMARS, 2020-02-25:
@jokezone - As I mentioned a few weeks back, I do not believe this to be a stop code in Windows. It may be a valid error log message and value, but not a stop code as in a kernel bug check that Windows created.  For general information on troubleshooting Windows with others I would recommend the WDK forum:

•	[Windows Hardware WDK and Driver Development](https://social.msdn.microsoft.com/Forums/windowsdesktop/en-US/home?forum=wdk)

closed 2020-02-25 by DOMARS

OsmiumBalloon, 2021-05-17:
Windows can absolutely generate a STOP error (blue screen) with this code.  Here are a few MSKB articles that mention it:

https://support.microsoft.com/en-us/topic/-0xc0000244-stop-error-when-the-security-log-reaches-its-maximum-size-in-windows-7-or-in-windows-server-2008-r2-50f4e4c9-8c50-64e7-7be8-9f0fbb581049

https://support.microsoft.com/en-us/topic/you-receive-a-stop-error-0xc0000244-audit-failed-error-message-during-the-shutdown-process-after-you-enable-the-crashonauditfail-setting-on-a-computer-that-is-running-windows-server-2008-or-windows-vista-dd0a1017-e550-7452-4d1d-f2e7c4b5d0c8

Here is a screenshot of a STOP crash I just induced, on Windows 10 LTSC 2019 (Win 10.1809, build 17763):

https://imgur.com/a/XyGsKCB

As noted, it is generated by Windows when `CrashOnAuditFail` is enabled, and Windows is unable to write an event to the Security log.  This is generally done in very-high-security environments where audit trail is paramount -- so if security logging is not working, it is seen as better to stop all work, than continue without logging.  It is controlled by Windows Security Option "Audit: Shut down system immediately if unable to log security audits".

There is something screwy about this code, though.  It does not appear in the [Bug Check Code Reference](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-code-reference2).  It does not appear in any copy of `bugcodes.h` I have checked.

`bugcodes.h` does define `AUDIT_FAILURE`, with a value of `0x86`, but I have never seen Windows generate a STOP with that value.

Further, a code with value `0xC0000244` does appear in `ntstatus.h`, with symbolic name `STATUS_AUDIT_FAILED`.

Perhaps the Windows internal implementation of the security log subsystem uses the `NTSTATUS` value, instead of the `BUGCODE` value, as an argument to the `KeBugCheck()` function?  That would arguably be a bug, but it should be documented regardless.

DOMARS, 2021-05-17:
@jokezone - 0xC0000244 may be a system error code that generates a bug check.  The system error codes are described here. 
https://docs.microsoft.com/en-us/windows/win32/debug/system-error-codes

And there is a defrag tools show (191) on this topic: 
https://www.youtube.com/watch?v=pRMEKKlJClE

If you have the !analyze output from the crash dump file it will list the stop code. 
```
kd> !analyze -v
*******************************************************************************
*                                                                             *
*                        Bugcheck Analysis                                    *
*                                                                             *
*******************************************************************************

DRIVER_IRQL_NOT_LESS_OR_EQUAL (d1)
```
!analyze is documented here: https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/using-the--analyze-extension