MicrosoftDocs/windows-driver-docs #1925: Windows cannot load the drivers

manker45, 2019-11-26:
What happens in the event that "Object name already exists" (Code31)? How can this problem be rectified?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 92e6e5fc-4c94-efd2-ddc1-a75da21b9e7f
* Version Independent ID: d38c2ff8-415c-7fa5-34ad-244bfab57d98
* Content: [Step 3 The Driver for the Device is Installed - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/step-3--the-driver-for-the-device-is-installed#feedback)
* Content Source: [windows-driver-docs-pr/install/step-3--the-driver-for-the-device-is-installed.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/install/step-3--the-driver-for-the-device-is-installed.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

RAJU2529, 2019-11-27:
@manker45. you can solve the problem by visiting Microsoft support official page which contains error code 31

**https://support.microsoft.com/en-us/help/310123/error-codes-in-device-manager-in-windows#bookmark-31**

manker45, 2019-11-27:
Thanks for your reply. I had already tried uninstalling/re-installing the device and the driver (several times) to no avail. At a suggestion from the device manufacturer, I did a search for 'remove non present devices from windows 10'. As suggested in the search results, I turned on 'View > Show hidden devices'  and found a non-present device. Once I uninstalled that and rebooted the machine the problem went away.

closed 2019-11-27 by manker45