MicrosoftDocs/windows-driver-docs #1926: invalid links

amethystAnt, 2019-11-27:
some of the links to function documentation don't work, for example ClosePort or InitializePortMonitor2. they lead to a page showing a 404 error message

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 6e551de7-c0b8-aeaf-da73-7bfdfe2ce647
* Version Independent ID: 2565c8ea-130e-a936-6e37-7f363738383b
* Content: [Port Monitor Server DLL Functions - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/print/port-monitor-server-dll-functions#feedback)
* Content Source: [windows-driver-docs-pr/print/port-monitor-server-dll-functions.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/print/port-monitor-server-dll-functions.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @barrygolden
* Microsoft Alias: **bagold**

RAJU2529, 2019-11-29:
@barrygolden. just now I created new PR #1931 
review those links, if those are correct then approve it

barrygolden, 2019-12-02:
@amethystAnt, I was not able to replicate the 404 error on those 2 pages. Please re-check to see if you are still receiving the 404 error. If so please post the full URL in a comment in this issue. Thanks!

barrygolden, 2021-01-27:
Closing as links are now working.

closed 2021-01-27 by barrygolden