MicrosoftDocs/windows-driver-docs #1929: The installer in the WDK folder after downloading is the Downloader itself.

babbsgit, 2019-11-29:
When I open wdksetup installer, it prompts me to download the wdk downloader again instead of installing the WDK.

RAJU2529, 2019-11-29:
@babbsgit If you already installed, it will only update new features or security or downloads a new version.
Which version of WDK, you trying to install

babbsgit, 2019-11-29:
@RAJU2529 I can't install it because of what I said above. Instead of installing the setup downloads the WDK setup again. The version I'm trying to install is WDK 10.018362.1.

My OS is windows 10.

babbsgit, 2019-11-30:
Try downloading it yourself:
https://go.microsoft.com/fwlink/?linkid=2085767

RAJU2529, 2019-11-30:
@babbsgit. I think I already installed same version on Windows 10 1909

EliotSeattle, 2019-12-02:
Hi @babbsgit 
Are you saving it locally before you try to run the executable?
Eliot

EliotSeattle, 2020-02-14:
Likely the answer to the last question was yes. Closing since there's no reply.

closed 2020-02-14 by EliotSeattle

BreakShoot, 2020-04-09:
What did you mean is he saving it locally? I am downloading it and this is the screen I get after the download. There was absolutely 0 point in the installation process is asked for me to install a VSIX plugin.

Before: https://i.imgur.com/cHE2HZ0.png
After: https://i.imgur.com/iSBSZo1.png

EliotSeattle, 2020-04-09:
Hi @BreakShoot 
Following your steps, if you then navigate to: C:\Program Files (x86)\Windows Kits\10
Is the wdksetup.exe file there for you to run?  You may have to save the exe to someplace other than Program Files, a folder which I cannot save directly to on my machine. 
I usually save the .exe to the desktop so I remember to delete it after it's installed.
Hope this helps,
Eliot

BreakShoot, 2020-04-27:
For other users that run across this. I'll note what fixed it. 

In one of the folders within the installation folder, there was a vsix file. I ran it, and it finally installed the plugin for me. 

There was no indication that I had to do this, and I had absolutely no luck trying to find this online. It literally took me checking every possible folder in the installation path until I found the vsix that looked like it should had been automatically ran after installation but wasn't.

EliotSeattle, 2020-04-28:
Sorry for the troubles, @BreakShoot 
I have passed this info to the WDK installation team to try and keep this from happening to others.
Eliot

colejhudson, 2020-06-05:
@BreakShoot Could you post the path to that vsix file, I've run into the same issue but can't seem to find it.

noxppw, 2020-06-09:
> 
> 
> For other users that run across this. I'll note what fixed it.
> 
> In one of the folders within the installation folder, there was a vsix file. I ran it, and it finally installed the plugin for me.
> 
> There was no indication that I had to do this, and I had absolutely no luck trying to find this online. It literally took me checking every possible folder in the installation path until I found the vsix that looked like it should had been automatically ran after installation but wasn't. I hope this helps someone, as this was an extremely frustrating experience.

Finally, after 2 days of searching your comment resolved my issue. Thx!

mfvmx, 2021-01-11:
> ~@BreakShoot Could you post the path to that vsix file, I've run into the same issue but can't seem to find that file. I see mention of `Vsix\VS2019\WDK.vsix` from `UserExperienceManifest.xml` but searching for it yields nothing.~
> 
> Ah ha! For posterity, per [this comment](https://github.com/appveyor/ci/issues/2994#issuecomment-504776140), the file is located in `C:\Program Files (x86)\Windows Kits\10\Vsix\VS2019`, which solved the issue for me as well.

This finally saved me, why is this not fixed yet?

ZakCheb, 2021-10-17:
> ~@BreakShoot Could you post the path to that vsix file, I've run into the same issue but can't seem to find that file. I see mention of `Vsix\VS2019\WDK.vsix` from `UserExperienceManifest.xml` but [...]

Thank you man, learning about kernel stuff is enough pain by itself, troubleshooting those obscure bugs on the way are super annoying.