MicrosoftDocs/windows-driver-docs #1934: Not possible to download WinDBG!

kjellegafvelin, 2019-12-03:
Why not offer a regular download of WinDBG?? It's not possible to download it when the IT Department has blocked Microsoft Store!!

---
#### Dokumentinformation

⚠ *Redigera inte det här avsnittet. Det krävs för docs.microsoft.com ➟ länkning till GitHub-problem.*

* ID: 8ea1adb7-1680-611f-a605-f8f10c7e26d8
* Version Independent ID: 37a27cc0-a7e9-adc0-2ce8-d9eb13df9c21
* Content: [WinDbg Preview - Installation - Windows drivers](https://docs.microsoft.com/sv-se/windows-hardware/drivers/debugger/windbg-install-preview#feedback)
* Content Source: [windows-driver-docs-pr/debugger/windbg-install-preview.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/windbg-install-preview.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

RAJU2529, 2019-12-03:
@kjellegafvelin.   i am trying to download the same app through Microsoft store. But failed

DOMARS, 2019-12-14:
Hello @kjellegafvelin, 

To make sure this request is logged in the right place, please use the feedback hub app in Windows.

There is also a current Git Issue on this as well, that you may want to comment on.

https://github.com/microsoftfeedback/WinDbg-Feedback/issues/19

We tend to focus on docs issues here.

closed 2019-12-14 by DOMARS