MicrosoftDocs/windows-driver-docs #1952: inf2cat fail for arm64 binaries

ttstanly, 2019-12-13:
Hi,

We met the inf2cat.exe(17763 version) fail on arm64 binaries. There are no warning and errors on the process but the final cat file is 0byte and got the message (Unable to get the length of the indirect data. Signability test failed.) from inf2cat in the end. We only see this failure on our build machine but my personal laptop is work for same binaries. Please see following output and give some advise. Thanks.

######################
Inf2Cat:
  C:\Program Files (x86)\Windows Kits\10\bin\10.0.17763.0\x86\inf2cat.exe /os:Server10_arm64 /VERBOSE /driver:C:\work\ARM64\Release\package\ 
  Processing directory (C:\work\ARM64\Release\package\) file (authenticatefam_securefp_arm64.dll)
  Processing directory (C:\work\ARM64\Release\package\) file (authenticatefam_securefp_ui_arm64.dll)
  Processing directory (C:\work\ARM64\Release\package\) file (synaefiresdll.dll)
  Processing directory (C:\work\ARM64\Release\package\) file (synafpadapter132.dll)
  Processing directory (C:\work\ARM64\Release\package\) file (synafpcoinstaller132.dll)
  Processing directory (C:\work\ARM64\Release\package\) file (synawudfbiousb.inf)
  Processing directory (C:\work\ARM64\Release\package\) file (synawudfbiousb132.dll)
  Processing directory (C:\work\ARM64\Release\package\) file (wbfresetservice132.exe)
  Parsing INF: C:\work\ARM64\Release\package\synawudfbiousb.inf
  Finished parsing INFs
  Processing INF: C:\work\ARM64\Release\package\synawudfbiousb.inf
  Finished processing INFs
  Testing driver package...
  Testing driver package...
  Testing driver package...
  Testing driver package...
  Testing driver package...
  Testing driver package...
  Testing driver package...
  Testing driver package...
  Testing driver package...
  Testing driver package...
  Testing driver package...
  Testing driver package...
  Testing driver package...
  Testing driver package...
  Testing driver package...
  Testing driver package...
  Testing driver package...
  Testing driver package...
  
  Signability test complete.
  
  Errors:
  None
  
  Warnings:
  None
  
  Unable to get the length of the indirect data.
  Signability test failed.
######################

Stanly

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 0afbc04b-349a-57ed-eeb0-3d2cd28e359c
* Version Independent ID: de61f684-31d7-2de5-d66c-a3319f5198d9
* Content: [Inf2Cat - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/devtest/inf2cat)
* Content Source: [windows-driver-docs-pr/devtest/inf2cat.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/devtest/inf2cat.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @EliotSeattle
* Microsoft Alias: **eliotgra**

RAJU2529, 2019-12-13:
@ttstanly . Will you send the packages in zip or .rar   compressed  formats

EliotSeattle, 2019-12-13:
Hi @ttstanly 
The Hardware Dev Center Support folks are the best bet to help investigate something like this.
1. **First**, sign into your hardware dev center account, and then 
2. Open a Hardware Dev Center support case for investigation. 
3. Go to 
[https://docs.microsoft.com/windows-hardware/drivers/dashboard/hardware-dashboard-faq](https://docs.microsoft.com/windows-hardware/drivers/dashboard/hardware-dashboard-faq)
and look for: 

**How do I contact Partner Center Support?**
If you are having problems accessing the Dashboard or need Dashboard Support please open a support ticket here: [https://developer.microsoft.com/windows/support](https://developer.microsoft.com/windows/support). Select **Contact us**, **Dashboard issue**, and then **Hardware submissions & signing (all OS version)** from the dropdown menu.

Hope this helps.
Eliot

closed 2019-12-13 by EliotSeattle

ttstanly, 2019-12-17:
@EliotSeattle   I take a look into the page you provided, it takes expensive money. Did you have other suggestion?

ttstanly, 2019-12-17:
> @ttstanly . Will you send the packages in zip or .rar compressed formats

attached the files here. I put the zip on the google driver.
https://drive.google.com/file/d/1ZXWMnfSHehjcFSJiNIq3rnKBD0RGQ8tU/view?usp=sharing

ttstanly, 2019-12-17:
> Hi @ttstanly
> The Hardware Dev Center Support folks are the best bet to help investigate something like this.
>
> 1. **First**, sign into your hardware dev center account, and then
> 2. Open a Hardware [...]

@EliotSeattle I take a look into the page you provided, it takes expensive money. Did you have other suggestion?

EliotSeattle, 2019-12-17:
Hi @ttstanly 

Your best bet is to use one or both of the two really good forums where other driver developers and Microsoft engineers can be found. The first is the MSDN Windows Hardware WDK and Driver Development forum: [https://social.msdn.microsoft.com/Forums/windowshardware/en-US/home?forum=wdk](https://social.msdn.microsoft.com/Forums/windowshardware/en-US/home?forum=wdk).

The other is the NTDEV forum from OSR Online: [https://community.osr.com/](https://community.osr.com/).
 
You can post your question there, and like-minded developers usually give feedback fairly quickly.
 
Again, thanks for reaching out. Best of luck with this issue.

ttstanly, 2019-12-18:
> Hi @ttstanly
>
> Your best bet is to use one or both of the two really good forums where other driver developers and Microsoft engineers can be found. The first is the MSDN Windows Hardware WDK and [...]

Thanks for the help and info! I will try with your advise.

Regards
Stanly