MicrosoftDocs/windows-driver-docs #1956: Publish wdk.vsix to the Visual Studio Marketplace

dickvdm, 2019-12-13:
This would rely help in deploying it through our organisation, as it then can be referenced in our Visual Studio extension as Extension Pack. Thanks a many!

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 5930505f-6810-b2a2-86b5-3c678aae7f89
* Version Independent ID: 28315cf8-f06f-23ad-2088-9d51d08efe0d
* Content: [Download the Windows Driver Kit (WDK) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/download-the-wdk#feedback)
* Content Source: [windows-driver-docs-pr/download-the-wdk.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/download-the-wdk.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2019-12-13:
Thanks, @dickvdm for the kind words!

EliotSeattle, 2019-12-13:
We'll see what we can do.

EliotSeattle, 2019-12-13:
Hi @dickvdm 
The WDK team has added your idea to their request mechanism and will examine need, cost, and feasibility. They and I thank you for the suggestion!!
Eliot

closed 2019-12-13 by EliotSeattle

dickvdm, 2019-12-16:
Hello,

Thanks for the follow up!

Please let me know when WDK.vsix is published to the Visual Studio Marketplace, so we can add it to our extension pack.

Thanks again!

With kind regards,

Dick van de Meulengraaf
Software Engineering | System Development 13

+31 (0)6 2904 8547
+31 (0)77 359 5940
dick.vandemeulengraaf@oce.com<mailto:dick.vandemeulengraaf@oce.com>

[cid:signature-NL2_ee17d3c6-df02-44be-84a6-ac43f1137e44.png]<https://www.oce.com/canon-announces-corporate-name-change-of-oce-to-canon-production-printing/>

Océ-Technologies B.V.
P.O. Box 101, 5900 MA Venlo
Sint Urbanusweg 43, 5914 CA Venlo
The Netherlands | www.oce.com<http://www.oce.com>
Trade register: 12002662

This message and attachment(s) are intended solely for use by the addressee and may contain information that is privileged, confidential or otherwise exempt from disclosure under applicable law. If you are not the intended recipient or agent thereof responsible for delivering this message to the intended recipient, you are hereby notified that any dissemination, distribution or copying of this communication is strictly prohibited. If you have received this communication in error, please notify the sender immediately by telephone or with a 'reply' message and delete the message and its attachment(s), if any, from your system(s). Thank you for your co-operation.

EliotSeattle, 2019-12-17:
Hi @dickvdm 
Just to set expectations, the engineering team will consider the request along with many others as part of their ongoing planning. I don't know if or when they will adopt the request, if ever. If they do, you will see it as an option on the [WDK Downloads](https://docs.microsoft.com/windows-hardware/drivers/download-the-wdk) page.

Eliot