MicrosoftDocs/windows-driver-docs #1958: messed-up tables here

landondyer, 2019-12-14:
Looks like there are three messed-up tables here (the ones starting with "HANDLE Event", and the one with "POINTER_32 Event" near the bottom).

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: cb0c1e81-8302-c76d-e2dc-add82d5e7eac
* Version Independent ID: 59e7f747-627c-edea-bedd-de4681971972
* Content: [Why Thunking Is Necessary - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/kernel/why-thunking-is-necessary#feedback)
* Content Source: [windows-driver-docs-pr/kernel/why-thunking-is-necessary.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/kernel/why-thunking-is-necessary.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

EliotSeattle, 2020-02-07:
Wow, THAT was fun with markdown. ;-)

Thanks for spotting this @landondyer and submitting the Issue. I've made the fix and create the PR. Should be live within 24 hours.

Eliot

closed 2020-02-07 by EliotSeattle

landondyer, 2020-02-07:
Thanks for fixing it. Loved your comment, my cow-orkers were amused :-)

tedhudek, 2020-02-07:
Completed PR in the private, should be live in a couple minutes

EliotSeattle, 2020-02-07:
Looks perfect on "review." Can go live. Thanks @tedhudek 

Glad to brighten up the workplace, @landondyer