MicrosoftDocs/windows-driver-docs #1960: "Windows 10 Preview SDK" doesn't exist

TheJCAB, 2019-12-17:
Or rather... in the current released build of VS (16.4.2), under "Universal Windows Platform development --&gt; Optional", there is no "Windows 10 Preview SDK". It's there only in Preview builds of VS2019.

If what the WDK needs is the "10.0.18362.0" build of the SDK (or higher?), it should just say so. In the current release of VS2019 it's (no longer) preview, instead "10.0.18362.0" is the SDK that's included (not optional) under "Universal Windows Platform development"

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 5930505f-6810-b2a2-86b5-3c678aae7f89
* Version Independent ID: 28315cf8-f06f-23ad-2088-9d51d08efe0d
* Content: [Download the Windows Driver Kit (WDK) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/download-the-wdk)
* Content Source: [windows-driver-docs-pr/download-the-wdk.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/download-the-wdk.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2020-01-06:
Researching

komsorg, 2020-04-30:
Proposed commit to fix this issue #2216

EliotSeattle, 2020-10-22:
Commit [2216](https://github.com/MicrosoftDocs/windows-driver-docs/pull/2216) was merged in May. Belatedly closing this issue.

closed 2020-10-22 by EliotSeattle