MicrosoftDocs/windows-driver-docs #1961: Scroll bar doesn't work well

deanmresource, 2019-12-18:
When browsing this page the left hand side menu list displays. This is correct.  This menu list is too big to fit on the page so it needs a scroll bar. However you have tried to be too clever and make it disappear when not in use. Unfortunately when it disappears it doesn't come back unless you start scrolling first. So in order to scroll the left hand side menu you have to put your mouse over the menu and start scrolling so that the scroll bar appears and you can then scroll the left hand menu. What idiot tested this and then signed it off as working correctly?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 85e7987a-ee73-57ac-05d3-60c045963022
* Version Independent ID: 87e5f7de-ab6c-2adc-b0ab-30ef1c0776f3
* Content: [Bug Check 0xA IRQL_NOT_LESS_OR_EQUAL - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-0xa--irql-not-less-or-equal)
* Content Source: [windows-driver-docs-pr/debugger/bug-check-0xa--irql-not-less-or-equal.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/bug-check-0xa--irql-not-less-or-equal.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2019-12-23:
@deanmresource - Not sure I follow here. Perhaps you describe the steps that you follow when the left menu disappears and what type of device is running Chrome?  

If the browser is made small enough at some point the left topic table of contents (TOC) menu is not rendered. But when the left TOC is present, I can scroll up and down in the TOC using the TOC control or I can scroll up and down in the topic using that scroll bar. I can click on either scroll bar to move around in either the TOC or the topic. 
Creating a web page that works across multiple form factors, languages and browsers is a challenge, but our current setup seems to work reasonably well in my and other team members, limited testing. But there may be ways that we should consider to improve how our content is presented.

deanmresource, 2019-12-24:
@DOMARS Hi yes i described the procedure to scroll. Basically, i can scroll the left hand TOC when i use the mouse scroll wheel. When i put the mouse over where the scroll bar should appear, it doesn't and i order to make it appear i have to scroll the mouse then "capture" the scroll bar with my mouse and then i can use that to scroll with.

This machine is windows 10 with chrome Version 78.0.3904.108 (Official Build) (64-bit)
there is an update waiting so i will apply that and see what my mileage is. This issue isn't a problem once you work out what you have to do in order to make the TOC scroll. You just can't put your mouse into the TOC area and make the scroll bar appear (which would be the correct actions).

DOMARS, 2019-12-26:
@deanmresource - Thanks for that clarification. As I understand it, the issue is that the scroll bar is not appearing when it should. I'm not able to replicate that.

I noticed a few things as I have looked at this:

- We seem to detect hover over for the mouse wheel to determine focus, with out requiring a click to bring focus. That is, without clicking on the topic or the TOC, if you hover over either, the mouse wheel will scroll in either of the two Windows.
- When the Window is so small that the TOC no longer is displayed, a "Contents" menu appears on the top right of the topic. 

Let me know how the update goes and if that changes the scroll bar rendering behavior.

deanmresource, 2020-01-06:
@DOMARS I have checked the scroll bar feature and it works excellently. The only thing i have noticed, which is of no consequence, is that the scroll bar doesn't disappear at all now however this allows someone to scroll the window at any time which removes the problem i saw before. This definitely works for me and i am sure others too.

Thanks

closed 2020-01-06 by deanmresource