MicrosoftDocs/windows-driver-docs #1963: Wrong value in inf sample

deryckk, 2019-12-20:
The value of PKEY_CompositeFX_Offload_StreamEffectClsid is "{D04E05A6-594B-4fb6-A80D-01AF5EED7D1D},19", not 11. Please correct this page.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 30d1680a-bb8d-1edc-4ed3-31621f966334
* Version Independent ID: 33068c61-dffb-04f9-511d-33faa42f05ef
* Content: [PKEY\_CompositeFX\_Offload\_StreamEffectClsid - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/audio/pkey-compositefx-offload-streameffectclsid#feedback)
* Content Source: [windows-driver-docs-pr/audio/pkey-compositefx-offload-streameffectclsid.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/audio/pkey-compositefx-offload-streameffectclsid.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

RAJU2529, 2019-12-21:
@deryckk , i verified with ComponentizedAudioSampleExtension.inx file , and i found  both 19 and 11  are  not correct , so  the correct value is  13

deryckk, 2019-12-21:
From audioenginebaseapo.h:
DEFINE_PROPERTYKEY(PKEY_CompositeFX_Offload_StreamEffectClsid, 0xD04E05A6, 0x594B, 0x4fb6, 0xA8, 0x0D, 0x01, 0xAF, 0x5E, 0xED, 0x7D, 0x1D, 19);

DOMARS, 2020-01-08:
@deryckk - This topic has been updated with the correct value of 19. Thanks for seeing this issue and thanks for creating this PR.

closed 2020-01-08 by DOMARS