MicrosoftDocs/windows-driver-docs #1966: download from public symbol server failed: connection refused

wyxloading, 2019-12-24:
I host a server to download windows system symbol (to analyze minidump) from Microsoft public symbol server.  
Recently some requests failed, sample details log below:
```
fail to req msdl symbol:https://msdl.microsoft.com/download/symbols/asycfilt.pdb/16A12E6932064CA2A4E5D5F5D8AC5B861/asycfilt.pdb
Get https://vsblobprodscussu5shard2.blob.core.windows.net/b-4712e0edc5a240eabf23330d7df68e77/A7E89DF4F90CECB95AFF8FB90E8AC7088A01B5CFE9439F56EF860E66649C35BD00.blob?sv=2017-04-17&sr=b&si=1&sig=ne6GzOiS3OUg76ITYxIykPabueNDtSU1mlXvRZIacis%3D&spr=https&se=2019-12-24T13%3A39%3A19Z&rscl=x-e2eid-caab545a-a4de4ad4-a4a59d79-b428f882-session-cbf50f70-b4d94ca1-a365b6a1-3c06aa71: dial tcp 104.214.40.16:443: connect: connection refused
```

more samples like:
```
https://msdl.microsoft.com/download/symbols/wdigest.pdb/2D67FDC36D8E435FB26C86EB9637FAB22/wdigest.pdb
https://msdl.microsoft.com/download/symbols/Flash.ocx/5BEDC1471c98000/Flash.ocx
https://msdl.microsoft.com/download/symbols/ucrtbase.pdb/90081979DD374FC0ACC20C2FC9EEED501/ucrtbase.pdb
...
```

DOMARS, 2020-01-06:
@Contagion21 - Any known issues right now with the symbol servers?

@wyxloading - Do you continue to receive the connection refused messages?

wyxloading, 2020-01-09:
Still get connection refused.

```
# wget 'https://msdl.microsoft.com/download/symbols/asycfilt.pdb/16A12E6932064CA2A4E5D5F5D8AC5B861/asycfilt.pdb'
--2020-01-09 09:57:05--  https://msdl.microsoft.com/download/symbols/asycfilt.pdb/16A12E6932064CA2A4E5D5F5D8AC5B861/asycfilt.pdb
Resolving msdl.microsoft.com (msdl.microsoft.com)... 204.79.197.219
Connecting to msdl.microsoft.com (msdl.microsoft.com)|204.79.197.219|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://vsblobprodscussu5shard2.blob.core.windows.net/b-4712e0edc5a240eabf23330d7df68e77/A7E89DF4F90CECB95AFF8FB90E8AC7088A01B5CFE9439F56EF860E66649C35BD00.blob?sv=2017-04-17&sr=b&si=1&sig=tQ02D6wjUGLdVJjA0GFHaBt79ADZUK2doIucQA8JpY0%3D&spr=https&se=2020-01-10T02%3A39%3A19Z&rscl=x-e2eid-9c9fef19-a35b4c44-a7fcdd4b-9950d674-session-5d0c7156-4b6b43d9-a0568858-2f3f7873 [following]
--2020-01-09 09:57:06--  https://vsblobprodscussu5shard2.blob.core.windows.net/b-4712e0edc5a240eabf23330d7df68e77/A7E89DF4F90CECB95AFF8FB90E8AC7088A01B5CFE9439F56EF860E66649C35BD00.blob?sv=2017-04-17&sr=b&si=1&sig=tQ02D6wjUGLdVJjA0GFHaBt79ADZUK2doIucQA8JpY0%3D&spr=https&se=2020-01-10T02%3A39%3A19Z&rscl=x-e2eid-9c9fef19-a35b4c44-a7fcdd4b-9950d674-session-5d0c7156-4b6b43d9-a0568858-2f3f7873
Resolving vsblobprodscussu5shard2.blob.core.windows.net (vsblobprodscussu5shard2.blob.core.windows.net)... 104.214.40.16
Connecting to vsblobprodscussu5shard2.blob.core.windows.net (vsblobprodscussu5shard2.blob.core.windows.net)|104.214.40.16|:443... failed: Connection refused.
```

DOMARS, 2020-01-23:
@wyxloading - I hope that you were able to determine your issue with the symbol server.

You can check this page for the current status of the Microsoft symbol server.

[Microsoft public symbol server](https://docs.microsoft.com/windows-hardware/drivers/debugger/microsoft-public-symbols)

For additional assistance related to using the Microsoft public symbol server I would recommend the Microsoft technical forums and OSR groups.

•	[Windows Hardware WDK and Driver Development](https://social.msdn.microsoft.com/Forums/windowsdesktop/en-US/home?forum=wdk)

•	[OSR's ntdev List](https://community.osr.com/categories/ntdev)

•	[OSR's windbg List](https://community.osr.com/categories/windbg)

Good luck in resolving your symbol server issues.

closed 2020-01-23 by DOMARS