MicrosoftDocs/windows-driver-docs #1967: OS Floor quesion

hatasum, 2019-12-25:
topic 1:
A floor is implicitly and automatically set when you select a Hardware ID and OS combination from the PnP grid. 
This means, the lowest OS you select from the PnP grid will be the implied floor.
topic 2:
If you select a value for the OS Floor that is earlier than the options listed in your PnP grid, you will receive the following error.

For example:
I select three HWIDs : RS3's HWID、RS4's HWID and 、19H1's HWID.
Q1 : By topic 1, the implied floor is RS3?
Q2 : Following Q1, but if I manual select OS floor to RS3, I will get the error by topic 2?

---
#### 文件詳細資料

⚠ *請勿編輯此區段。 這是 docs.microsoft.com ➟ GitHub 問題連結的必要項。*

* ID: e888e07f-1323-59a9-2c2f-14e9f814f552
* Version Independent ID: de2aa427-004a-b1a8-270d-cf13d0c9ed91
* Content: [How to limit or expand a driver’s distribution based on Windows version - Windows drivers](https://docs.microsoft.com/zh-tw/windows-hardware/drivers/dashboard/limit-driver-distribution#important-information)
* Content Source: [windows-driver-docs-pr/dashboard/limit-driver-distribution.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/dashboard/limit-driver-distribution.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @EliotSeattle
* Microsoft Alias: **eliotgra**

tedhudek, 2020-11-10:
Hello @hatasum, thanks for the feedback and apologies for the delay.

To help us troubleshoot, is there supposed to be an error in your report when you say "you will receive the following error"?

hatasum, 2020-11-11:
@tedhudek , thanks for your reply.
the error is: Flooring OS should be greater than the pnp selection OS. 
During this period, I have understood the cause of this error, I will close this issue, thank you.

closed 2020-11-11 by hatasum

tedhudek, 2020-11-11:
Thank you, @hatasum.

Do you think we need to add anything to the doc page to help others who might encounter this error?

hatasum, 2020-11-11:
@tedhudek ,
"Flooring OS should be greater than the pnp selection OS"
Maybe doc page can add more clear detail(or example) to explain it.

tedhudek, 2020-11-11:
Hi @hatasum, I am not familiar with this feature.  Do you remember how you fixed it?