MicrosoftDocs/windows-driver-docs #1971: Verifier Command

jbrown3859, 2020-01-02:
On this page, it says that the command to verify drivers is Verifer. It is actually Verifier.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 7f81b449-5ccf-0633-bb19-f391d612476e
* Version Independent ID: 59109407-b187-25e3-9b0f-392ac0370c86
* Content: [Bug Check 0x50 PAGE_FAULT_IN_NONPAGED_AREA - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-0x50--page-fault-in-nonpaged-area#feedback)
* Content Source: [windows-driver-docs-pr/debugger/bug-check-0x50--page-fault-in-nonpaged-area.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/bug-check-0x50--page-fault-in-nonpaged-area.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

RAJU2529, 2020-01-02:
@illfated.  Please review the  PR #1973

DOMARS, 2020-01-02:
Thanks everyone for seeing this issue and creating a PR to fix it.

closed 2020-01-02 by DOMARS