MicrosoftDocs/windows-driver-docs #1972: Can UCSI client driver supports PPM external? And How to implement if it supported?

CuongDV17, 2020-01-02:
Hi, Currently I'm developing UCSI client driver following the document below.
As I understand, currently it only supports for internal PPM. Could I know if it also supports for external PPM? and how to implement to support for external PPM?

Thanks and best regards,
D.V.Cuong

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: b1b26c19-4f96-d00e-da07-a5e13bfb74ae
* Version Independent ID: fa8743ee-2cbf-3dd0-4f62-26893de1171b
* Content: [Write a UCSI client driver - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/usbcon/write-a-ucsi-driver#feedback)
* Content Source: [windows-driver-docs-pr/usbcon/write-a-ucsi-driver.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/usbcon/write-a-ucsi-driver.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @billlattimer
* Microsoft Alias: **billla**

CuongDV17, 2020-02-10:
Hi @billlattimer 
Please let me know if you have any updates on this? Thank you!

tedhudek, 2021-03-07:
Closing out since the user who opened it deleted their account and we have no way of reaching them.  Please re-open if needed

closed 2021-03-07 by tedhudek