MicrosoftDocs/windows-driver-docs #1978: Automating Disabling Driver Signature Enforcement

itsTreble, 2020-01-09:
So is there no way to disable driver signature enforcement anymore?  This affects CI with windows...

None of the following work:
BCDEdit /set LoadOptions DDISABLE_INTEGRITY_CHECKS 
BCDEdit /set TESTSIGNING ON 
BCDEdit /set nointegritychecks ON
REG ADD "HKCU\Software\Policies\Microsoft\Windows NT\Driver Signing" /v BehaviorOnFailedVerify /t REG_DWORD /d 0 /f

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 3aeb487d-b749-4f6e-9ff5-71be68bf7a09
* Version Independent ID: 44418645-8c6e-7387-ba3e-02f478816961
* Content: [Installing an Unsigned Driver during Development and Test - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/installing-an-unsigned-driver-during-development-and-test)
* Content Source: [windows-driver-docs-pr/install/installing-an-unsigned-driver-during-development-and-test.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/install/installing-an-unsigned-driver-during-development-and-test.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

RAJU2529, 2020-01-10:
@itsTreble . You can disable driver signature enforcement  using  F8 button during boot 

the commands work fine, but before entering the command you must disable Secure Boot or UEFI boot in BIOS.

commands  should work in administrator rights

BCDEdit.exe /set LoadOptions DISABLE_INTEGRITY_CHECKS
BCDEdit.exe /set TESTSIGNING ON
BCDEdit.exe /set nointegritychecks ON
BCDEdit.exe /set {bootmgr} flightsigning off
BCDEdit.exe /set flightsigning off

I have  own experienced on my  laptop and  tried on college desktops.

tedhudek, 2020-07-17:
Thanks for the feedback, and apologies for the delay.  I don't have deep experience with test signing, but what you have there looks like the still supported path.

You can find more details on these pages:

* [Enable Loading of Test Signed Drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/the-testsigning-boot-configuration-option)
* [Test Signing](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/test-signing)

If that fails, you may have success posting your question in the following forums:

* [Windows Hardware WDK and Driver Development](https://social.msdn.microsoft.com/Forums/windowsdesktop/en-US/home?forum=wdk)
* [OSR's ntdev List](http://www.osronline.com/showlists.cfm?list=ntdev)

closed 2020-07-17 by tedhudek

RetroSoftwareRepository, 2022-09-21:
Still can't get a system I built recently with an old Marvell wireless card to start without disabling driver signature enforcement from the advanced startup menu. Clearly Microsoft broke this permanent workaround to get my wireless card working all the time. It's a Netgear WG311v3 from the mid 2000's.