MicrosoftDocs/windows-driver-docs #1980: 1st parameter 0x12 missing

MrPepka, 2020-01-09:
While debugging the blue screen 0x13A I came across the first parameter 0x12 which is not described on this page. I am asking for a description of this parameter

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 25edc27f-390e-725f-7ee7-638a3cbeb5bf
* Version Independent ID: baaa985d-e78a-ff2d-5fb8-ea9e63358eff
* Content: [Bug Check 0x13A KERNEL_MODE_HEAP_CORRUPTION - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-0x13a--kernel-mode-heap-corruption)
* Content Source: [windows-driver-docs-pr/debugger/bug-check-0x13a--kernel-mode-heap-corruption.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/bug-check-0x13a--kernel-mode-heap-corruption.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

TheBestYTech, 2020-01-25:
Same. Happened when testing bug check 13a on not my fault, with kernel debugger connected.

Grettonista, 2020-02-10:
Ist parameter 0x12. All I can tell is that dxgmm2.sys faulted. No idea why.

DOMARS, 2020-02-14:
@MrPepka - The parameter list has been updated to include the more recent parameter ones such as 0x12. Thanks for this issue, and your interest in improving the stop code documentation.

closed 2020-02-14 by DOMARS