MicrosoftDocs/windows-driver-docs #1982: Is there any special right granted by 'group ownership'?

kyanha, 2020-01-11:
From what I'm reading, the owner of a file always has "change security", even when otherwise removed or denied.

Is there any special permission that's always granted to the group which owns a file?  If not, it might be useful to explain that it's only really used and assigned when inheriting ACLs which contain ACEs for CREATOR_GROUP (SID S-1-3-1) [assuming that I understand its use correctly -- if I don't, then I haven't been able to figure out what it's used for].

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: f59cc28c-947c-e661-03f1-41027669af88
* Version Independent ID: a3234bb3-d6b8-47e0-0d10-b19cc7a14bbb
* Content: [Security Descriptors - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ifs/security-descriptors)
* Content Source: [windows-driver-docs-pr/ifs/security-descriptors.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/ifs/security-descriptors.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @lorihollasch
* Microsoft Alias: **loriwhip**

deftclaw, 2020-05-07:
Nothing "special", beyond that each user within the group would then have "Change Security", "even when otherwise removed or denied" (except now removing a user from said group will effectively cancel this) 

That can be useful for its own purposes, suppose a user transfers departments but they are still a manager of permissions over certain groups, now you just move them to gardening_owners_group from cashiers_owners_group and they lose "change security" over cashier_files and gain "change security" over gardening_files, without ever explicitly changing the user's permissions.(so you don't have to adjust each user's granular permissions but instead manage which permissions-based group they belong to)

deftclaw, 2020-05-07:
Sorry "this" is ambiguous, effectively cancel the removed-user's "Change Security" permissions, while leaving the Group's permissions in-tact

lorihollasch, 2020-10-20:
@kyanha , sorry for my late response to this issue! I'm closing it since @deftclaw answered your question (thanks, @deftclaw !), and since there isn't anything to add to the device driver documentation. The SDK has lots of information about DACLs - [this page](https://docs.microsoft.com/en-us/windows/win32/secauthz/dacls-and-aces) seems relevant to your question.

closed 2020-10-20 by lorihollasch