MicrosoftDocs/windows-driver-docs #1984: Guidance on how to find other registry keys that depend the removed filter driver?

HenkPoley, 2020-01-11:
Just a little remark, if I use the PowerShell oneliner to uninstall 'hpdskflt' on my macbook, this driver remains 'in use' in other places. So at the next reboot Windows goes into a BSOD because it can't access the harddrive, to which you have to do recovery to get it to a working state.

Is there any guidance how to clean up the remaining pieces that keep the driver 'in use' while it is no longer there ?

---

The OS drive has been in an HP laptop, that's when it picked up this disk filter driver, for parking the harddrive heads at a sudden drop, that I no longer need:

* Driver name: hpdskflt.sys
* Description: HP Disk Filter - SATA/RAID
* Version: 4.2.9.1
* Product name: Mobile Data Protection System

This driver attaches itself to the main disk drive as a filter driver.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: c05d6a6d-434c-538a-d000-3a96236e6bff
* Version Independent ID: 3314e31f-5c06-4694-f40a-c29692681c4e
* Content: [Using an INF File to Uninstall a File System Filter Driver - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ifs/using-an-inf-file-to-uninstall-a-file-system-filter-driver#feedback)
* Content Source: [windows-driver-docs-pr/ifs/using-an-inf-file-to-uninstall-a-file-system-filter-driver.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/ifs/using-an-inf-file-to-uninstall-a-file-system-filter-driver.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @lorihollasch
* Microsoft Alias: **loriwhip**

RAJU2529, 2020-01-12:
if your Laptop or desktop running with Windows OS , then the file is located in 
C:\windows\system32\drivers\

find hpdskfit.sys  , take ownership of this file and folder drivers

once completed ownership, you can delete it

HenkPoley, 2020-01-12:
Yeah, and then your computer won't boot 😂

RAJU2529, 2020-01-12:
@HenkPoley . try with same driver with previous version.

HenkPoley, 2020-01-12:
I don't need this driver in any version on my current laptop. 

Does Microsoft have a tool that takes an INF file, and tells you what files and registry keys get installed?

RAJU2529, 2020-01-12:
@HenkPoley . then contact the HP customer care for help.   We cant help you for Windows issues for individual users. 
so post your queries answers.microsoft.com

HenkPoley, 2020-01-12:
Thanks for the answers, I guess I'll have to write such a tool myself.