MicrosoftDocs/windows-driver-docs #1985: [Question] Driver Dynamic Update

andresfererra45, 2020-01-12:
I developed a software product that uses a kernel driver, however, some of my users have been complaining about having to update the application by themselves. I have used pop-ups to prompt them to update, but I'd like to make it simpler.
Can I change the setting to include my driver in Windows Updates for automatic distribution in new major Windows releases? Also, will old installations get the update as well - meaning those who already installed the software will also be able to benefit from the automatic updates?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 531ef6ee-7a29-e6d0-56d6-e08a923b0aa4
* Version Independent ID: d00f4b65-2ae6-643f-d67a-c04f916a7058
* Content: [Publish a driver to Windows Update - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/publish-a-driver-to-windows-update#feedback)
* Content Source: [windows-driver-docs-pr/dashboard/publish-a-driver-to-windows-update.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/dashboard/publish-a-driver-to-windows-update.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @EliotSeattle
* Microsoft Alias: **eliotgra**

RAJU2529, 2020-01-12:
@andresfererra45 . yes you can  add it.
and also your kernel driver must be uploaded to Microsoft server so that users can update via windows update if compatible driver is found

tedhudek, 2020-11-11:
Hi @andresfererra45 

Thanks for the feedback and apologies for the delay getting back to you.

We recommend that you republish (not resubmit) the driver in Hardware Dev Center as **Automatic**.

If you want the driver to come down automatically only during OS upgrades, select only the first checkbox shown on https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/understanding-windows-update-automatic-and-optional-rules-for-driver-distribution.

If you want everyone to get the driver as soon as it is available, select both checkboxes.

To republish to Windows 10 and Windows Server 2016/2019 only, go to the shipping label and change the option from **Manual** to **Automatic**.

Hope this helps!

closed 2020-11-11 by tedhudek