MicrosoftDocs/windows-driver-docs #1991: How to configure the modem log  to be saved as a file in Windows?

moolwalk, 2020-01-14:
This document page doesn't explain how the modem log can be saved?
Can the modem log be consumed by ETW session? 
Or does WWanSvc saves the log in any special session or local file?
Does user application have to handle the modem log using MBIM command?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 9a49c708-4a65-da81-5e4a-79887a5082d5
* Version Independent ID: b59ce8c9-0a81-f15e-2991-e7f66cdb0648
* Content: [MB modem logging with DSS - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/network/mb-modem-logging-with-dss#feedback)
* Content Source: [windows-driver-docs-pr/network/mb-modem-logging-with-dss.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/network/mb-modem-logging-with-dss.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @billlattimer
* Microsoft Alias: **billla**

nogarder77, 2020-02-04:
Also, Command code of MBIM_CID_MODEM_LOGGING_CONFIG is not defined.

tedhudek, 2021-03-03:
Hi @moolwalk, thanks for the feedback and apologies for the delay.

There is no modem in the market which has implemented modem logging support using `MBIM_CID_MODEM_LOGGING_CONFIG` and the corresponding DSS data channel.
If you are planning for your modem to support this CID, please let us know so we can best support you.

>This document page doesn't explain how the modem log can be saved?

We do not currently provide support for the end user to control modem logging behavior.

>Can the modem log be consumed by ETW session?

Not at present – the modem log is in binary format; not in string format.

>Or does WWanSvc save the log in any special session or local file?

Wwansvc does not save this log in any special session. While it's not supported currently, it's possible to enable this functionality via netsh. We can either make the logs available via ETW session or dumped as a file from class-driver or even made available as part of WinRT layer.

>Does user application have to handle the modem log using MBIM command?

If application has the capability of invoking MBIM CIDs or corresponding OIDs, the modem log data can be made available for the application. Please refer to the flow diagram available here --

https://docs.microsoft.com/windows-hardware/drivers/network/mb-modem-logging-with-dss

After configuring the logging parameters with `MBIM_CID_MODEM_LOGGING_CONFIG`, the app would open the DeviceServiceSession data channel with `MBIM_CID_DSS_CONNECT`.

Once the modem writes any logs to the newly created data channel, the modem calls [**MbbDeviceReceiveDeviceServiceSessionData**](https://docs.microsoft.com/windows-hardware/drivers/ddi/mbbcx/nf-mbbcx-mbbdevicereceivedeviceservicesessiondata), the data from which is available to apps via the WinRT layer : [**MobileBroadbandDeviceService**](https://docs.microsoft.com/uwp/api/windows.networking.networkoperators.mobilebroadbanddeviceservice).

closed 2021-03-04 by tedhudek