MicrosoftDocs/windows-driver-docs #1993: Clarification regarding IPv6 hash types

I-gor-C, 2020-01-15:
> NDIS_HASH_TCP_IPV6
> If this flag alone is set, the NIC should parse the received data to identify an IPv6 packet that contains a TCP segment. The NIC must identify and skip over any IPv6 extension headers that are present in the packet. If the NIC cannot skip over any IPv6 extension headers, it should not calculate a hash value.
> 

There is no mentioning of handling fragmented IPv6 packets (i.e. with Fragment extension header). In case such packet arrives, it's obviously can't contain a TCP segment (which is a requirement from the quote above). For IPv4 the documentation explicitly mentions the required handling of fragmented packets, yet nothing is defined for IPv6 fragmented packets.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: ebcc9a73-4aff-5bd0-191e-4704ad77e909
* Version Independent ID: ec8a4a15-8c84-89b6-973d-99aa0b2bdb4e
* Content: [RSS Hashing Types - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/network/rss-hashing-types)
* Content Source: [windows-driver-docs-pr/network/rss-hashing-types.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/network/rss-hashing-types.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @billlattimer
* Microsoft Alias: **billla**

tedhudek, 2021-02-27:
Internal PR underway

tedhudek, 2021-03-02:
Updated in b59719bd5d.  Thanks for the heads-up!

closed 2021-03-02 by tedhudek