MicrosoftDocs/windows-driver-docs #1995: Windows Server 2019

sonvoba, 2020-01-15:
Running "WDK Test Target Setup x64-x64_en-us.msi" on a Windows Server 2019 target system does not produce outcomes described in this article.  For example, it does not create the "C:\DriverTest" folder.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: ce050ea3-d61e-830c-3038-f725eb4c29e1
* Version Independent ID: eb7fc65e-bcfe-dc86-3f8c-336d20eb0c86
* Content: [Provision a computer for driver deployment and testing (WDK 10) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/gettingstarted/provision-a-target-computer-wdk-8-1#feedback)
* Content Source: [windows-driver-docs-pr/gettingstarted/provision-a-target-computer-wdk-8-1.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/gettingstarted/provision-a-target-computer-wdk-8-1.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @EliotSeattle
* Microsoft Alias: **eliotgra**

DOMARS, 2020-01-15:
@sonvoba - We can check with the dev team on Windows Server 2019.

As another option, you may want to create and INF file and install the driver manually on the target machine as described in the step by step echo debugger lab: https://docs.microsoft.com/windows-hardware/drivers/debugger/debug-universal-drivers---step-by-step-lab--echo-kernel-mode-

sonvoba, 2020-01-16:
Thank you for the tip.  I’ll give the manual option a try.

 

Regards,

 

--svb

DOMARS, 2020-01-23:
Following up with the dev team, so far no response.

RAJU2529, 2020-01-29:
this article says the example   C:\DriverTest only.  not the actual folder

DOMARS, 2020-11-17:
@sonvoba as @RAJU2529 mentions, the text says, indicates the directory is an example, so I think this reference is OK. Personally, I prefer to use the process  to manually deploy the driver using and INF file, as that is a workflow that the driver will typically need to support at some point in the development process. Thanks for this issue, and good luck with your driver development.

closed 2020-11-17 by DOMARS