MicrosoftDocs/windows-driver-docs #1997: Latency for isochronous transfers with the USB Driver stack

paamand, 2020-01-16:
The USB Driver stack has a limitation for isochronous transfers: `the number of isochronous packets per frame is calculated as 8 / polling period`.
This means that it is impossible to use microframes for isochronous transfers and severely limits the roundtrip latency that can be achieved in latency critical applications.

Is there any way around this limitation?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: bb022c50-a63b-00cf-d6e9-a31b226b4412
* Version Independent ID: 9d930cf8-7610-1f4b-06b5-f8c70e400a38
* Content: [Best Practices - Using URBs - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/usbcon/usb-client-driver-contract-in-windows-8#do-not-use-polling-period-greater-than-8-for-high-speed-and-superspeed-isochronous-transfers)
* Content Source: [windows-driver-docs-pr/usbcon/usb-client-driver-contract-in-windows-8.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/usbcon/usb-client-driver-contract-in-windows-8.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @billlattimer
* Microsoft Alias: **billla**

tedhudek, 2021-03-02:
Tracking in internal bug

tedhudek, 2021-03-11:
Hi @paamand thanks for the feedback and apologies it's taken us so long.  As I mentioned above, your question is now tracked in a Microsoft-internal bug.  I do not have an ETA for when it will be resolved, so I am going to close out this issue.

If you would like me to notify you if and when I get an answer, please just reply to me here and I will see it and make a note.

Thanks again for caring about the docs!

closed 2021-03-11 by tedhudek