MicrosoftDocs/windows-driver-docs #1998: Driver not working on some windows 10 after Attestation

sumit79, 2020-01-16:
Hi,

We followed the process and it did worked well. Now if we install this driver on windows 10 laptops and desktops. Some of the windows 10 laptop giving error 39 while others are working good. 

What can be possible issue. 

Thanks
SM

EliotSeattle, 2020-01-16:
Hi @sumit79 
Can you please tell me the URL for the page you are referencing?

sumit79, 2020-01-17:
Hi @EliotSeattle 

I am following this 

https://techcommunity.microsoft.com/t5/windows-hardware-certification/driver-signing-changes-in-windows-10-version-1607/ba-p/364894

Then created a new device in windows hardware portal and signed it. 

Thanks

EliotSeattle, 2020-04-23:
Hi @sumit79 

I am so sorry that I let this sit. This feedback forum gets sent to the Windows driver documentation team. We use it to track ***documentation*** issues. The Hardware Dev Center Support folks are the right people to help investigate a customer problem with submissions, like this. To open a ticket with the support team:

1. **First**, sign into your hardware dev center account, and *then*...
2. Go to [Hardware Dashboard FAQ](https://docs.microsoft.com/windows-hardware/drivers/dashboard/hardware-dashboard-faq), where you'll find this passage: 
 
**How do I contact Partner Center Support?**
If you are having problems accessing the Dashboard or need Dashboard Support please open a support ticket here: [https://developer.microsoft.com/windows/support](https://developer.microsoft.com/windows/support). Select **Contact us**, **Dashboard issue**, and then **Hardware submissions & signing (all OS version)** from the dropdown menu.

It's important that you sign into your account first.

If support, working with the dashboard developers, finds that something needs to change in the docs, we'll then update that here. I hope this helps. I am terribly sorry that I did not get you this info immediately.

Eliot

closed 2020-04-23 by EliotSeattle