MicrosoftDocs/windows-driver-docs #2001: DigiCert SysDev page is not longer available

alfaunits, 2020-01-19:
Now the links shoots straight to the order page, with high prices.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 93d3f097-d81b-66c2-4835-d63758687d34
* Version Independent ID: fe9d1cdd-966e-89b5-c66b-60531b0ec655
* Content: [Get a code signing certificate - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/get-a-code-signing-certificate)
* Content Source: [windows-driver-docs-pr/dashboard/get-a-code-signing-certificate.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/dashboard/get-a-code-signing-certificate.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @EliotSeattle
* Microsoft Alias: **eliotgra**

EliotSeattle, 2020-01-21:
Hi @alfaunits 

The Partner Center for Windows Hardware enables you to create and manage your system and driver submissions quickly and easily. It replaces the legacy (Sysdev) dashboard for hardware tasks including: hardware certification, collaborative driver development, and driver distribution through Windows Update.

Hope this helps.

Eliot

alfaunits, 2020-01-21:
Hi. No, I mesnt the link for "Buy a DigiCert EV code signing certificate" is no longer working. Instead of their "Sysdev friends" page, it sends us to the order cart directly.

EliotSeattle, 2020-01-21:
Ah, OK. I will have to do a little hunting for that answer.
Thanks!

cubiclesoft, 2020-04-08:
DigiCert removed the SysDev route.  Now they only produce super-expensive certificates.

Microsoft needs to completely redo the code signing platform so it is affordable for everyone.  Certs are now just far too expensive for ISVs and SMBs, especially internationally (most currencies are far more diluted than the USD).  DigiCert's SysDev reduced pricing was the only barely sane thing in the industry.  After DigiCert was mostly bought out in 2019, people started complaining online about DigiCert's code signing program dramatically went down in quality.  DigiCert support is apparently not what it used to be.  And then this year they had the audacity to raise the price without improving the quality of the service!

There's something rather wrong when someone can buy a used car every couple of years for the same price as a few KB of data.  And it's not like the companies issuing code signing certs are actually doing anything special.  They all pretty much look exclusively at D&B to determine if a business is "legitimate" but D&B is hardly a trustworthy organization.  Everyone and their mom can check D&B in 5 minutes - and the software developers working for these companies have automated the entire process and outsourced support to Umblekistanizan.  There's absolutely no way an EV code signing cert is worth anywhere close to $700+ USD per _year_.  It's one thing to charge a steep first-time account setup fee - it's another to charge the exact same fee year-over-year when nothing extra is done to warrant the cost and provide sub-par product support.  In addition, all of the code signing certificate companies take money with the user's expectation that they will issue a certificate right away but the companies refuse to return any of the money they took in the event they decide to not issue the certificate.  Taking money for a service and not rendering said service is, by definition, fraud.  The fraud I just described happens on a daily basis and is, of course, illegal.

Microsoft can do a LOT better than align themselves with these companies.  You literally own GitHub.  There's DNSSEC DANE TLSA.  There's Azure.  There are also plenty of other technologies that have come about since the inception of Authenticode.  There's got to be a way to accomplish the same goals that Authenticode originally set out to accomplish but for far less cost.

jtraulle, 2020-04-09:
Could not agree more with you @cubiclesoft.
Racket, extortion ... you name it.

If there is no more an affordable alternative, I will just fall back to gpg signing + hashes.

It is unbelievable that a company like Microsoft could not propose something fair (even Apple manage to do this with their Developer Program that includes certificates for $99 a year).

EliotSeattle, 2020-04-09:
Thanks, @cubiclesoft and @jtraulle for the comments and feedback. 

I've forwarded your feedback to the right people, those who can track or act on your concerns. My team handles only the documentation for the process, which seems to accurate in the link. Those who make the process now have your comments.

All the best.
Eliot

closed 2020-04-09 by EliotSeattle