MicrosoftDocs/windows-driver-docs #2002: Procedure does not yield windbg on VS 2017

aucampia, 2020-01-20:
I am using VS 2017, and I only have option of Installed Windows 10 SDK (specific version) or not - I don't have sub options of debug tools. Please fix this procedure so it is correct for how to get windbg from Visual Studio or make it clear that it is not possible if it is not.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 162fe2fe-9216-f4c5-9f91-3df8776bb99e
* Version Independent ID: 013dbf6b-0790-9c11-24f4-dbf24d64401e
* Content: [Download Debugging Tools for Windows - WinDbg - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/debugger-download-tools#feedback)
* Content Source: [windows-driver-docs-pr/debugger/debugger-download-tools.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/debugger-download-tools.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2020-01-23:
@aucampia - Thanks for this input. For information on debugging in Visual Studio see [Visual Studio debugger documentation](https://docs.microsoft.com/en-us/visualstudio/debugger/?view=vs-2017). Windbg is a separate set of tools that is not available from Visual Studio.  Was there a specific docs topic that indicated that the WinDbg tools were available in Visual Studio? As there are two debugging tools sets, I can see how this could be confusing.

aucampia, 2020-01-24:
@DOMARS from [here](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/debugger-download-tools) ([source](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/debugger-download-tools.md))

> ## [image] Debugging Tools for Windows 10 (WinDbg)

> If you just need the Debugging Tools for Windows 10, and not Windows Driver Kit (WDK) for Windows 10 or Visual Studio 2017, you can install the debugging tools as a standalone component from the Windows SDK. In the SDK installation wizard, select **Debugging Tools for Windows**, and deselect all other components.

> - Get Debugging Tools for Windows (WinDbg) from the SDK: [Windows 10 SDK](https://developer.microsoft.com/windows/downloads/windows-10-sdk).

Following this link takes me [here](https://developer.microsoft.com/en-us/windows/downloads/windows-10-sdk) and this says:

> There are two ways to get the Windows 10 SDK. You can install it from this web site, by selecting the download link, or you can by select this version of the Windows 10 SDK (10.0.18362.0) in the Visual Studio 2019 Installer optional components.

So I install Visual Studio (2017) in my case, select Windows 10 SDK - but afterwards I don't have WinDbg

DOMARS, 2020-05-02:
@aucampia - Thanks for the clarification. I added additional wording to indicate wording to indicate that the Windows Debugging Tools are not available as part of Visual Studio.

closed 2020-05-02 by DOMARS