MicrosoftDocs/windows-driver-docs #2008: Missing closing brace

wumb0, 2020-01-23:
if (SUCCEEDED(hr) {
should be
if (SUCCEEDED(hr)) {

Right above the "Retrieve enumerator for interfaces of the binding path." comment.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 53a3f100-6c13-4bb4-2e20-277dfc6f625d
* Version Independent ID: 8ccd7c9f-95d8-9c54-1be3-7e1526570914
* Content: [Changing Bindings for a Component - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/network/changing-bindings-for-a-component)
* Content Source: [windows-driver-docs-pr/network/changing-bindings-for-a-component.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/network/changing-bindings-for-a-component.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @billlattimer
* Microsoft Alias: **billla**

RAJU2529, 2020-02-15:
@illfated. Please review this PR #2042.

billlattimer, 2020-02-28:
@wumb0 , @RAJU2529 thank you for finding this on this issue and #2042. Fixed and merged.

closed 2020-02-28 by billlattimer

illfated, 2020-02-28:
interesting to see that you commit the fixes manually without merging the Pull Requests, in contrast to most other repositories here on GitHub.

billlattimer, 2020-03-02:
@illfated I already had fixes for the article, so I combined those and pushed. 

In terms of the duplicate GitHub issue on this, I marked that as a dupe and closed it out. 

Thanks!

illfated, 2020-03-02:
Very well, thanks for enlightening me on the situation. Fair enough, that's good enough for me. :)