MicrosoftDocs/windows-driver-docs #2017: Outdated information

phuclv90, 2020-01-28:
In this page it's said that the user/kernel address space is 8TB/248TB

&gt; For a 64-bit process, the virtual address space is the 8-terabyte range 0x000'00000000 through 0x7FF'FFFFFFFF.

&gt; ...

&gt; The 8-terabyte range from 0x000'00000000 through 0x7FF'FFFFFFFF is used for user space, and portions of the 248-terabyte range from 0xFFFF0800'00000000 through 0xFFFFFFFF'FFFFFFFF are used for system space.

However in [Memory Limits for Windows and Windows Server Releases](https://docs.microsoft.com/en-us/windows/win32/memory/memory-limits-for-windows-releases) I see that since Windows 8.1 and Windows Server 2012 R2 the user/kernel address space has been changed to 128/128 TB

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: ebe87d57-198f-66f4-69ff-6fdbc15eb214
* Version Independent ID: 880a9327-3f4f-83eb-c0f7-030588344330
* Content: [Virtual address spaces - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/gettingstarted/virtual-address-spaces)
* Content Source: [windows-driver-docs-pr/gettingstarted/virtual-address-spaces.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/gettingstarted/virtual-address-spaces.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @EliotSeattle
* Microsoft Alias: **eliotgra**

EliotSeattle, 2020-01-28:
@tedhudek assigning this and #1878 to you, as they deal with kernel info. Can you please look at these to see what updates are needed in the topic? Thanks.

tedhudek, 2020-02-11:
Thanks for the heads-up.  A fix will appear in the next few minutes.

closed 2020-02-11 by tedhudek

BRabbit777, 2020-04-23:
@tedhudek It looks like there is additional out of date info on the page. Down midway under "User Space and System Space" there is a diagram and explanatory blurb that still reference the out of date 8TiB User address space, and 248TiB system address space.

tedhudek, 2020-04-23:
Hi @BRabbit777 , I appreciate the feedback.  I went ahead and removed the sentence and diagram.

If you like, I'd encourage you to submit a pull request if there are particular points/links you'd like to see added to this page.  To do that, click the **Edit** button on the docs.microsoft.com page, and then click the pencil ("edit this file") icon on GitHub.  Happy to take a look at anything you think would be worthwhile.