MicrosoftDocs/windows-driver-docs #2021: Does PTP inbox driver support DFx?

bill-chien, 2020-02-03:
I executed a single-device test on the Precision Touchpad (PTP) with inbox driver. I tested the PTP inbox driver of OS 19H1 and OS build 19041, but both results were failed. The notebook that I tested supported the Modern-Standby feature, and another audio device driver in this notebook passed DFx single-device test. Could you tell me whether the PTP inbox driver supports DFx?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 53456e5d-2f98-0b55-dfb0-dcb25c4eb7f9
* Version Independent ID: 3ca90a7c-0bf7-7711-9235-cb423cc11764
* Content: [Introduction to the Directed Power Management Framework - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/kernel/introduction-to-the-directed-power-management-framework)
* Content Source: [windows-driver-docs-pr/kernel/introduction-to-the-directed-power-management-framework.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/kernel/introduction-to-the-directed-power-management-framework.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

chcurlet-msft, 2020-02-15:
I'm not familiar with the PTP inbox driver. It is not one of the inbox drivers that has added DFx support. Is it on the DRIPS constraint path on your system?

bill-chien, 2020-02-18:
Hi chcurlet-msft,
I used wpr command to generate a .etl file and used WPA to check the device list. PTP device belongs to non-PoFx category, so I think PTP device is not DRIPS constraint.

chcurlet-msft, 2020-02-20:
non-PoFx devices can still have constraints, so you may want to double check by checking “Microsoft-Windows-Kernel-Power.DeviceVerboseRundown” event from the etl. If the field PlatformStateDependents is not 0, then the device has a constraint.  (If it's 0, then it's not a constraint.)

bill-chien, 2020-02-20:
Thanks chcurlet. I look into the Microsoft-Windows-Kernel-Power.DeviceVerboseRundown event. PTP device doesn't show in the device list but its parent device 'Intel I2C Serial I/O Host Controller' does and its PlatformStateDependents is 1. According to the guide document, starting in 20H1, DFx is required for all drivers of devices that are DRIPS constraints or children of constraints, except for the following Microsoft drivers: NDIS, HIDClass, Bluetooth, and Storport. Even though PTP device's parent device is DRIPS constraint, PTP is an exception because its driver is HIDClass. As a result, PTP driver is not DRIPS constraint.

chcurlet-msft, 2020-02-20:
Correct, if HIDClass is its power policy owner (PPO) then it is exempted by default.

closed 2020-02-24 by tedhudek